ksheumaker/homeassistant-apsystems_ecur #272: Unable to get correct data from ECU, and no cached data. See log for details, and try power cycling the ECU.

defaliz, 2024-10-04:
Hi,
thanks for the work...
is it possible to restart remotely ecu-b with your integration ?
today i've no data, so i restart HA, and now i've this message : 
Unable to get correct data from ECU, and no cached data. See log for details, and try power cycling the ECU. 

Ema app and web are OK, i'm not near the ecu-b but far away, no possibility to restart Ecu-B ?
regards

defaliz, 2024-10-05:
Ok today intégration works fine.  .. maybe we must wait for the sun rise ?

HAEdwin, 2024-10-07:
Indeed might depend on the time of day to immediate get results. There is no method to restart the ECU-B remotely from within this integration. HA is build for automating so fortunately you can create an automation by using a smart-plug and using the ECU query device switch. I'll close this issue because I think it is solved now. If you have any more questions, feel free to open another or use the discussions section.

closed 2024-10-07 by HAEdwin