ksheumaker/homeassistant-apsystems_ecur #273: Add "Zero Export" for ECU-C devices?

hautamik, 2024-10-06:
I have recently installed solar panel system using APSystems' ECU-C. Most of my smallish home automation is done with Node Red, but I'm keen to start using Home Assistant. I found this ECU-related component which seems to be usable with my system, but I am hoping I could use HA to set my ECU-C's Zero Export when electricity price goes below zero.

Unfortunately my experience in programming dates from 1980's coding (in COBOL), so I'm not easily able to fork and start adding my own code. I was able to find out that ECU-C can be controlled with certain URL and HTTP POST method to set and unset Zero Export.

Using curl:

```shell
curl --location 'http://[ECU-C IP ADDRESS]/index.php/meter/set_meter_display_funcs' \
--header 'Content-Type: application/x-www-form-urlencoded' \
--data-urlencode 'meter_func=1' \
--data-urlencode 'this_func=1' \
--data-urlencode 'power_limit=0'
``` 

'meter_func=1' sets Zero Export on, and 'meter_func=0' sets it off. Power_limit takes the power in Watts, so to restrict outbound energy to 1 kW, one should 'use power_limit=1000'.

From the web page source I could find out that 0 (zero) as response means success. I guess some other response is a failure.

Does this make any sense, could it be possible to implement this feature with given information?

HellStorm666, 2024-11-22:
I would love this option aswell.

marcello7512, 2025-01-16:
As far as I know zero export is only available with CT coils. I have now the ECU-C running in HA an trying to set something up zo ik can change the values of the QS1 inverters via http://<ecu-c-ip>/index.php/configuration/maxpower

HAEdwin, 2025-01-29:
@HellStorm666 @hautamik 
I might add this option to the new integration at: https://github.com/HAEdwin/homeassistant-apsystems_ecu_reader
Also, with this new version you should be able to switch individual inverters off/on depending on needs.
I do not own an ECU-C so I hope it works! Please let me know at: https://github.com/HAEdwin/homeassistant-apsystems_ecu_reader

I'll close this issue for now, feel free to remind me at the mentioned git when you still really want the feature.

closed 2025-01-29 by HAEdwin