ksheumaker/homeassistant-apsystems_ecur #274: Install failed (unknown error occured)

Dark-Raver, 2024-10-18:
Since some time I´ve the problem to connect to the ECU-B again. Month´s ago it works without Problems.
What I did: Deleted the device in the device manager, uninstalled it from HACS, reinstalled a fresh new version.

By adding the "APSystems PV solar ECU" device I always receive the error message "Unknown error occured" - see attached screenshot. The IP-Adress is correct and I can also ping it via terminal (HA) and terminal from Windows.

HA was restarted several times, the ECU was disconnected from power several times (reboot).
I have no more ideas, what I can do for now.
[image]

HAEdwin, 2024-10-18:
Does the home-assistant.log show any error messages?

Dark-Raver, 2024-10-19:
> Does the home-assistant.log show any error messages?

Here the Logfile:
2024-10-18 22:38:34.221 ERROR (MainThread) [custom_components.apsystems_ecur.config_flow] APSystemsInvalidData exception: [Errno 111] Connection refused
Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 91, in open_socket
    self.sock.connect((self.ipaddr, self.port))
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/config/custom_components/apsystems_ecur/config_flow.py", line 42, in async_step_user
    test_query = await self.hass.async_add_executor_job(ap_ecu.query_ecu)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 98, in query_ecu
    self.open_socket()
  File "/config/custom_components/apsystems_ecur/APSystemsSocket.py", line 94, in open_socket
    raise APSystemsInvalidData(err)
custom_components.apsystems_ecur.APSystemsSocket.APSystemsInvalidData: [Errno 111] Connection refused

After sleeping one night I tried again this morning by adding new device and typing in IP in required field -> Connected without any problems. Nothing was changed during last night. So I am a little bit wondering,

HAEdwin, 2024-10-19:
In most cases it's the ECU that's in a wrong state, inverters are offline or whatever else with the hardware. Reinstalling the integration does not help unless you've made modifications to it. Great it is solved. I'll close this issue. Feel free to open another when needed.

closed 2024-10-19 by HAEdwin