ksheumaker/homeassistant-apsystems_ecur #275: No data after night - have to restart home assistant

christianvaes, 2024-10-20:
Hi, I'm using this integration from the beginning with my ecu-r 2160(firmware ECU_R_1.3.14). However, pretty often the connection with the ecu doesn't come up in the morning after the night when the solar panels didn't report/produced anything. When I restart home assistant it all works fine, so I think there is nothing wrong with the ecu itself(I never have to power cycle it to get it to work).

What can I do to fix this? Restarting Home Assistant every day doesn't sound like a very good solution.

HAEdwin, 2024-10-20:
Weird... let's start by enabling debugging in the integration and see what the home-assistant.log comes up with. What is the status of the ECU Query Device switch in the morning?

christianvaes, 2024-10-20:
Hi, i've enabled debug loggin now, let's see when it happens again. I will share logs when it doesn't come up again.

[image]
This is the status this morning, so seems it's off. After a restart it turns on again.

HAEdwin, 2024-10-20:
I guess we can skip the debugging in that case. My advise would be to always add the ECU Query Device entity somewhere on the dashboard.
[image]

Check the settings and you might want to increase the "Retries when ECU fails" interval.
[image]

Check the switch on the dashboard in the morning and turn it back on, you don't have to restart HA.

christianvaes, 2024-10-20:
Hi, thank you. I think I tried this switch in the past but remembered it didn't work. But I will try again next time it happens. But I still would like to have an automated solution.

Does it make sense to create an automation to enable the switch in case it's off and the sun comes up? 
This integration doesn't do anything without sunlight anyway I think. So maybe it would be a nice improvement to have a feature to disable querying after sundown(with some offset) and enable with sunset again?

My settings are:
[image]

I think the retries is already above yours, so is it recommended to increase further? I'm also doubtful why this only happens during the night, do you have a similar behavior?

HAEdwin, 2024-10-21:
A nice feature of Home Assistant is to create automations. In you case all the tools you need are present to create an automation. In Yaml it would look something like:
```
alias: ECU query off 1 hour after sunset
description: ""
triggers:
  - trigger: sun
    event: sunset
    offset: "+01:00:00"
conditions: []
actions:
  - type: turn_off
    device_id: 6aaxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
    entity_id: ab7xxxxxxxxxxxxxxxxxxxxxxxxxxxxx
    domain: switch
mode: single
```
For this you'll need the sun integration which is here: https://www.home-assistant.io/integrations/sun

HAEdwin, 2024-10-21:
> I think the retries is already above yours, so is it recommended to increase further? I'm also doubtful why this only happens during the night, do you have a similar behavior?

There's no problem in increasing the value, try what works best. I do not have similar behavior and I'm querying my 2160 all night. It's weird that situations differ that much, my only guess is that there is much more data maintenance going on at your site after sunset which results in conflicts when querying. It would mean that data is not being received well at EMA during the day.

HAEdwin, 2024-10-21:
In my case, if the query switch is turned off (that happens once in a while) I use a smart plug from the Action to hardware reset the ECU with an automation.
```
alias: Powercycle_ECU
description: Reset the ECU when query switch turns off
mode: single
triggers:
  - entity_id:
      - switch.ecu_query_device
    from: "on"
    to: "off"
    trigger: state
conditions: []
actions:
  - data:
      message: ECU restart
    action: notify.mobile_app_iphone_xxxxxxx
  - type: turn_off
    device_id: f7fexxxxxxxxxxxxxxxxxxxxxxxxxxx
    entity_id: 75cxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
    domain: switch
  - delay:
      hours: 0
      minutes: 0
      seconds: 20
      milliseconds: 0
  - type: turn_on
    device_id: f7fexxxxxxxxxxxxxxxxxxxxxxxxxxxxx
    entity_id: 75cxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
    domain: switch
  - delay:
      hours: 0
      minutes: 1
      seconds: 0
      milliseconds: 0
  - data: {}
    target:
      entity_id: switch.ecu_query_device
    action: switch.turn_on
```

HAEdwin, 2024-10-28:
I'm closing this issue assuming it is resolved. If there is still an issue, feel free to open a new issue.

closed 2024-10-28 by HAEdwin

christianvaes, 2024-10-31:
For a week it seems to work now, i didn't implement a restart yet so will have  a look when it happens again. Thank you for all the help.

y-pinky, 2025-07-01:
Hi,

I just wanted to let you both know that christianvaes is not the only one with this issue - it also happens on my installation.

Symptom: after the night, no new data is being collected by HA. Restarting HA fixes the issue but cannot be the resolution.

The workaround presented in this thread (switch on "ECU Query Device" if it is off) works. It is possible to do this using automation, e.g. "When ECU Query Device witches to off for a minute, then switch it back on".

The question remains, why does "ECU Query Device" switch off, and why does it not automatically switch on in the next morning?