ksheumaker/homeassistant-apsystems_ecur #278: Exception error: can only concatenate str (not "NoneType") to str

gartcimore, 2024-11-26:
After upgrading to Home Assistant Core 2024.11.3 I got the following in logs (after reboot, and all the time since) : 

```
This error originated from a custom integration.

Logger: custom_components.apsystems_ecur
Source: custom_components/apsystems_ecur/__init__.py:144
integration: APSystems PV solar ECU (documentation, issues)
First occurred: 8:45:02 AM (220 occurrences)
Last logged: 1:48:36 PM

Using cached data from last successful communication from ECU. Exception error: can only concatenate str (not "NoneType") to str
```

gartcimore, 2024-11-26:
> Did you allready reboot the ECU?

I did not, never had the need to do that in previous updates. Rolled back to before my updates and the integration is still not working and I had local access to the ECU

After power cycling it, still the same error

gartcimore, 2024-11-26:
After another set of power cycling and refresh, everything seems to work correctly. 

Sorry for the noise and thanks for all the work for this

closed 2024-11-26 by gartcimore

HAEdwin, 2024-11-27:
No problem, unless APsystems is careless with their firmware releases I don't expect the functionality to suddenly disappear.