ksheumaker/homeassistant-apsystems_ecur #280: Error integration

kaka83136, 2025-01-01:
Hello,

For over a month, the integration has not worked anymore.

I tried to reinstall the integration but I have this error message when adding: Unknown error occurred

[image]

How to make it work again?

I have an ECU-C

Thanks

tonejay62, 2025-01-03:
Hi, Got the same problem since yesterday... My ECU-C is online i can get data from web interface...
[image]

kaka83136, 2025-01-03:
Hello,
Just reboot the ECU.
I was able to reintegrate it to HA

HAEdwin, 2025-01-11:
Thanks for the feedback, I'll close this issue. Feel free to open a new one if needed.

closed 2025-01-11 by HAEdwin