ksheumaker/homeassistant-apsystems_ecur #281: Connection frequenstly lost since HA update January 2025

Joern-HAP, 2025-01-25:
Hello,

integration worked fine over years now, but since Junuary Iam losing the data multiple times a day. Then I need to restart HA Server to get data again. How/ where can I log the to understand the problem?
Netcat worked as described in installation & I provided fix IP.

Joern-HAP, 2025-01-25:
After restarting HA -sometimes even:

Dieser Fehler stammt von einer benutzerdefinierten Integration

Logger: custom_components.apsystems_ecur
Quelle: custom_components/apsystems_ecur/__init__.py:139
Integration: APSystems PV solar ECU (Dokumentation, Probleme)
Erstmals aufgetreten: 11:08:17 (7 Vorkommnisse)
Zuletzt protokolliert: 11:09:32

Using cached data from last successful communication from ECU. Invalid data error: [Errno 111] Connection refused

HAEdwin, 2025-01-25:
The reason why you need to restart HA is that you are overlooking the Query Device Switch. Place this switch on the dashboard, you only have to turn it on to see results again. A restart of HA does the same thing. 

I now own an ECU-R 2160 and ECU-R-Pro 2162 and cannot reproduce problems like you describe. Only occasionally does the query fail. Which factors exactly influence this; poor WiFi connection, interference on the ECU, who knows, but the placement of the ECU is certainly an important factor. Try to place the ECU in a more favorable location.

HAEdwin, 2025-01-29:
@Joern-HAP I'll close this issue for now, take a look at the new version at: https://github.com/HAEdwin/homeassistant-apsystems_ecu_reader also. Feel free to open a new issue when needed.

closed 2025-01-29 by HAEdwin