ksheumaker/homeassistant-apsystems_ecur #285: Error connecting to ECU-R (ImportError: cannot import name 'STATE_CLASS_MEASUREMENT)

Roald28, 2025-02-19:
Hello,

Since I think update 2025 (it is not working for a while now) I cannot connect anymore. I did a few power cycles with the ECU-r. Also did a reinstallation in Hacs to make sure I have te latest and correct verion.

It gives me the error below. So it does see the ECU-R and the inverters that are connected to the ECU-R. But it cannot connect. Can someone point me in the right direction because I am stuck at this moment. 

ImportError: cannot import name 'STATE_CLASS_MEASUREMENT' from 'homeassistant.components.sensor'

HAEdwin, 2025-02-19:
The latest ECU firmware does not deliver complete ECU and inverter data after a reboot so I have to adapt to that. May take a while because I also have to do software tests. Meanwhile make sure you wait at least 10 minutes after reboot  to enable the ECU to recover (get inverter data) and deliver complete and valid data again. After that you will again be able configure and to connect.

Roald28, 2025-02-19:
Hi HAEdwin, Thanks for the quick response. If I understand correctly in my case it does not work at all. And when it is working I should give the system 10 minutes before pulling the data?

I needed for testing just send me een message, happy to try some beta releases for debugging.

HAEdwin, 2025-02-19:
Yes, give the ECU 10 minutes to recover from the restart/reboot and then you can do the configuration or later on configuration changes. It's just the 10 minutes after reboot in which the ECU contains invalid data. After that, data will be valid for querying over time. If this solved the issue, feel free to close it. I will continue further development at: https://github.com/haedwin/homeassistant-apsystems_ecu_reader