ksheumaker/homeassistant-apsystems_ecur #288: "Einrichtungsfehler" coulndt get Entities from ECU - B

mitkoester, 2025-03-16:
Hy,

for a longer time i couldn't get Enteties from my ECU-B. 
The ECU as well as the Inverter are found as devices, but both do not have any entities.
I also tried the other Integration "[homeassistant-apsystems_ecu_proxy]" This Integration couldn't find any devices or entities.
The System worked with the old integration last year. i cann't say til when the integration did not work.

In the protocol i can find these Problems:

"Logger: homeassistant.config_entries
Quelle: config_entries.py:753
Erstmals aufgetreten: 14:55:38 (2 Vorkommnisse)
Zuletzt protokolliert: 14:56:18
Error setting up entry ECU: 216300036310 for apsystems_ecur

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 753, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/apsystems_ecur/__init__.py", line 210, in async_setup_entry
    await hass.config_entries.async_forward_entry_setups(config, PLATFORMS)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2560, in async_forward_entry_setups
    await integration.async_get_platforms(platforms)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1123, in async_get_platforms
    if platform := self._get_platform_cached_or_raise(platform_name):
                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1214, in _get_platform_cached_or_raise
    raise ModuleNotFoundError(
    ...<2 lines>...
    )
ModuleNotFoundError: Platform apsystems_ecur.binary_sensor not found"

"Logger: homeassistant.config_entries
Quelle: config_entries.py:753
Erstmals aufgetreten: 12:04:42 (1 Vorkommnisse)
Zuletzt protokolliert: 12:04:42
Error setting up entry ECU: 216300036310 for apsystems_ecur

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 753, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/apsystems_ecur/__init__.py", line 210, in async_setup_entry
    await hass.config_entries.async_forward_entry_setups(config, PLATFORMS)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 2560, in async_forward_entry_setups
    await integration.async_get_platforms(platforms)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1184, in async_get_platforms
    import_future.result()
    ~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1184, in async_get_platforms
    import_future.result()
    ~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1184, in async_get_platforms
    import_future.result()
    ~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1154, in async_get_platforms
    await self.hass.async_add_import_executor_job(
        self._load_platforms, platform_names
    )
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1097, in _load_platforms
    platform_name: self._load_platform(platform_name)
                   ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1271, in _load_platform
    cache[full_name] = self._import_platform(platform_name)
                       ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1303, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1324, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'custom_components.apsystems_ecur.binary_sensor'"

For the other Integration i can find this Problem:

"Logger: homeassistant.config_entries
Quelle: config_entries.py:967
Erstmals aufgetreten: 11:58:57 (1 Vorkommnisse)
Zuletzt protokolliert: 11:58:57
Error unloading entry APsystems ECU proxy for apsystems_ecu_proxy

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 967, in async_unload
    result = await component.async_unload_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/apsystems_ecu_proxy/__init__.py", line 85, in async_unload_entry
    await api_handler.async_shutdown()
  File "/config/custom_components/apsystems_ecu_proxy/__init__.py", line 167, in async_shutdown
    await self.midnight_tracker_unregister()
TypeError: object NoneType can't be used in 'await' expression"

HAEdwin, 2025-03-16:
Looks like your integration installation is not compleet `ModuleNotFoundError: No module named 'custom_components.apsystems_ecur.binary_sensor'"` somehow this file is missing, you might have deleted it or did the install wrong.

I'd recommend https://github.com/HAEdwin/homeassistant-apsystems_ecu_reader  Read the installation instructions and use HACS.

mitkoester, 2025-03-16:
thanks for your fast answer.

i de- and reistalled the integration. now it works again. 
i tried it before but i think i forget to restart HA before reinstalling.

HAEdwin, 2025-03-16:
@mitkoester Glad it worked for you, I was in a bit of a hurry. Uninstalling should have been a first advice of course. If you have any more questions, I welcome you @ https://github.com/HAEdwin/homeassistant-apsystems_ecu_reader and don't forget to star my repository. Thanks!

closed 2025-03-16 by HAEdwin