ksheumaker/homeassistant-apsystems_ecur #289: Apsystems_ecur' sets option flow config_entry explicitly, which is deprecated

sebastianfisahn, 2025-03-19:
Hello,

i found this warning in my logs today. Just for awareness:

Logger: homeassistant.helpers.frame
Quelle: helpers/frame.py:324
Erstmals aufgetreten: 11:29:52 (1 Vorkommnisse)
Zuletzt protokolliert: 11:29:52
Detected that custom integration 'apsystems_ecur' sets option flow config_entry explicitly, which is deprecated at custom_components/apsystems_ecur/config_flow.py, line 64: self.config_entry = config_entry. 
**This will stop working in Home Assistant 2025.12**, 
please create a bug report at https://github.com/ksheumaker/homeassistant-apsystems_ecur/issues

Best Sebastian

sebastianfisahn, 2025-03-19:
> HAEdwin
>  left a comment
> (ksheumaker/homeassistant-apsystems_ecur#289)
> Hi @sebastianfisahn , this git is no longer being maintained by me. In the newer version at https://github.com/HAEdwin/homeassistant-apsystems_ecu_reader this issue does not occur.

Hello Edwin,

Thx for the prompt response! I was aware of the new integration but not aware of that this needs to be manually switched instead of being migrated automatically.

I switched now and it runs like a charm.

Thx for your amazing work.

Best Sebastian

elz64, 2025-06-21:
Intégration suddenly faiied. Maybe After today's HA update to ECU: 215000072758
[2025.6.1]
No ECU are now replying all entities are 'dead'
Don't see I log anything relevant 
APC apps is still running right ...

ECU: 215000072758
[5 appareils](https://4cx4l65alz4buhvbikugl6pqdyl8kryn.ui.nabu.casa/config/devices/dashboard?historyBack=1&config_entry=01JE5X3VPD1FEBARY5XGT8B46D) et [33 entités](https://4cx4l65alz4buhvbikugl6pqdyl8kryn.ui.nabu.casa/config/entities?historyBack=1&config_entry=01JE5X3VPD1FEBARY5XGT8B46D)
Échec de la configuration, nouvel essai: Unable to get correct data from ECU, and no cached data. See

HAEdwin, 2025-06-22:
Hi @elz64 Please upgrade to the latest version at https://github.com/HAEdwin/homeassistant-apsystems_ecu_reader